konnected-io/konnected-esphome #4: Incorrect device_class for 'WiFi Signal %' Entity

othorp, 2023-03-24:
Running Home Assistant 2023.3.6 with Konnected ESP8266 template: https://github.com/konnected-io/konnected-esphome/blob/master/alarm-panel-esp8266.yaml

Home Assistant log is reporting:
`Entity sensor.alarm_panel_wifi_signal_2 (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']`

Looking at: packages/wifi.yaml@master seems it may be necessary to override inherited device_class for the WiFi Signal % entity?

closed 2023-07-03 by heythisisnate

othorp, 2023-07-04:
This doesn't seem to be fixed.  Home Assistant log is still reporting errors following commit #66:

`Entity sensor.alarm_panel_wifi_signal_2 (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'Signal %' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dBm', 'dB']`

heythisisnate, 2023-07-09:
I don't think this needs to be changed as the message is just a warning, and in my opinion, the `signal_strength` device class still makes the most sense here, despite not supporting `%` as a unit.

If you want, you can override the device class by adding something like this to your config:
```
sensor:
  - id: !extend wifi_signal_pct
    device_class: battery
```
This was taken almost verbatim from this [ESPHome example](https://esphome.io/components/sensor/wifi_signal.html) so I don't think there's an actual problem in adding a sensor to represent WiFi signal as a percentage.

atx32, 2023-07-10:
This same issue is being discussed in https://github.com/home-assistant/core/issues/86053 so hopefully a resolution will come out of it.

commit 8ebb7ba6 (2023-07-03): fixes #4; update unit of measure for WiFi Signal %