konnected-io/konnected-esphome #7: alarm panel pro keeps appearing as adoptable despite being configured

daggy72, 2023-05-03:
despite having successfully installed and configured my AlarmPanelPro in ESPHome the module keeps appearing in ESPHome Panel as a new, adoptable, device... 

[image]

below is the config, did I miss something?
```
esphome:
  name: konnected-alarm-panel-pro
  platform: ESP32
  board: wesp32

#use ETHERNET instead of WIFI
ethernet:           
  type: LAN8720
  mdc_pin: GPIO16
  mdio_pin: GPIO17
  clk_mode: GPIO0_IN
  phy_addr: 0
#wifi:
#  ssid: !secret wifi_ssid_not
#  password: !secret wifi_password_not
  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Konnected-ESPHome Hotspot"
#    password: !secret wifi_password_not

# MQTT
#mqtt:
#  broker: 192.168.10.12
#  username: !secret mqtt_username
#  password: !secret mqtt_password

# NATIVE API for HOME ASSISTANT
# Enables the native API for Home Assistant
# more: https://esphome.io/components/api.html
api:

# OTA UPDATES
# Enables over-the-air updates
# more: https://esphome.io/components/ota.html
ota:
  password: !secret ota_password

####
# WEB SEVER
# Enables the built-in web server for viewing the device state, internals and controls via web browser
# on the same local network as the device.
web_server:
  include_internal: true

#LOGGING
logger:

binary_sensor:
  - platform: gpio
    name: Zone 1
    device_class: door
    pin:
      number: GPIO4
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 2
    device_class: door
    pin:
      number: GPIO2
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 3
    device_class: window
    pin:
      number: GPIO15
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 4
    device_class: window
    pin:
      number: GPIO13
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 5
    device_class: motion
    pin:
      number: GPIO18
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 6
    device_class: motion
    pin:
      number: GPIO14
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 7
    pin:
      number: GPIO33
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 8
    pin:
      number: GPIO32
      mode: INPUT_PULLUP

  - platform: gpio
    name: Zone 9
    pin:
      number: GPIO36
      mode: INPUT

  - platform: gpio
    name: Zone 10
    pin:
      number: GPIO39
      mode: INPUT

  - platform: gpio
    name: Zone 11
    pin:
      number: GPIO34
      mode: INPUT

  - platform: gpio
    name: Zone 12
    pin:
      number: GPIO35
      mode: INPUT

switch: 
  - platform: gpio
    name: Siren
    pin: GPIO12 # ALARM1

  - platform: gpio
    name: Strobe
    pin: GPIO5  # ALARM2

  - platform: gpio
    name: Status LED
    pin: GPIO3

  # with a piezo buzzer, beep 3 times
  - platform: gpio
    pin: GPIO23 # OUT1
    id: beep
  - platform: template
    name: Beep beep beep
    turn_on_action:
      - switch.turn_on: beep
      - delay: 60ms
      - switch.turn_off: beep
      - delay: 60ms
      - switch.turn_on: beep
      - delay: 60ms
      - switch.turn_off: beep
      - delay: 60ms
      - switch.turn_on: beep
      - delay: 60ms
      - switch.turn_off: beep
```

daggy72, 2023-05-03:
and suddenly, after a few yaml configuration updates and reboots of HA it has dissapeared .... 
so we can close it...

closed 2023-05-03 by daggy72

heythisisnate, 2023-05-03:
@daggy72 this is caused by a change in the `name` attribute of the ESPHome device. When first discovering using our pre-built firmware, the unique id of the device is automatically added to the device `name`. However when you adopt it and customize the YAML, the unique ID is not preserved. If you want the device name and hostname to stay the same, just add back the unique ID into the YAML file `name` attribute.