konnected-io/konnected-esphome #8: "incorrect_codes" counter is not updated in local alarm

jordan875, 2023-05-10:
When using local alarm, the `incorrect_codes` counter is not updated when failed_code is triggered. Therefore, the alarm is not triggered when the incorrect codes count is exceeded.

jordan875, 2023-05-12:
Fixed by adding transitions to the failed_code input.

Changed from: 
`inputs:
...
    - name: failed_code
      on_input:
        - lambda: id(incorrect_codes) += 1;
        - if:
            condition:
              lambda: return id(incorrect_codes) >= 3;
            then:
              - state_machine.transition: trigger
 `
 
Changed to: 
`inputs:
...
    - name: failed_code
      transitions:
        - armed_away -> triggered
        - armed_home -> triggered
        - armed_night -> triggered
      on_input:
        - lambda: id(incorrect_codes) += 1;
        - if:
            condition:
              lambda: return id(incorrect_codes) >= 3;
            then:
              - state_machine.transition: trigger
 `

diogopms, 2023-11-07:
@jordan875 did you figure out a solution for this? thanks