konnected-io/konnected-esphome #12: Latest ESPHome reserves id 'alarm', causing build errors

depoll, 2023-07-05:
ESPHome 2023.6.4 has started erroring on builds with the following:

```
ID 'alarm' is reserved internally and cannot be used.
  id: alarm
  name: Alarm
  pin: D8
  platform: gpio
  icon: mdi:bullhorn
  internal: true
```

heythisisnate, 2023-07-05:
Thank you for reporting. This caught me by surprise and will, unfortunately, require a breaking change on our side.
Related to: https://github.com/esphome/esphome/pull/5042

I think the simple fix is to rename this id to `siren`. Any other thoughts?

si458, 2023-07-09:
just picked up my esp8266 and same issue here
```
INFO ESPHome 2023.6.4
INFO Reading configuration /config/esphome/alarm-panel-6a4ee8.yaml...
Failed config

switch.gpio: [source /config/esphome/.esphome/packages/d9505d5d/packages/alarm-panel/alarm.yaml:3]
  
  ID 'alarm' is reserved internally and cannot be used.
  id: alarm
  name: Alarm
  pin: D8
  platform: gpio
  icon: mdi:bullhorn
```

heythisisnate, 2023-07-09:
@si458 you can customize the features you want by editing the config in ESPHome. Replace the default package with the list of packages you need. See: https://support.konnected.io/esphome-on-the-konnected-alarm-panel-esp8266

si458, 2023-07-09:
thanks @heythisisnate!
i looked all over the support page and didnt find that link anywhere!
it was hidden under the 'see more' option of the esphome section!

commit 3634ee5b (2023-07-09): Renames  to  to work around new ESPHome reserved ID. Fixes #12
closed 2023-07-09 by heythisisnate