konnected-io/konnected-esphome #15: Adopting Alarm Panel Pro 1.8 breaks ethernet

d4l3k, 2023-08-25:
Hi, I recently broke ethernet on my device by clicking the adopt button in ESPHome. Documenting the issue so other can fix it.

The default config when adopting uses the LAN8720 config rather than the RTL8201 config needed by Alarm Panel Pro 1.8.

```
lan87xx: lan87xx_init(499): wrong chip ID
```

Updating the esphome configuration to use:

```
  konnected.alarm-panel-pro-esp32-wifi: github://konnected-io/konnected-esphome/release/alarm-panel-pro-esp32-rtl8201.yaml@master
```
instead of the generic one on master resolves the issue.

wjwnz, 2023-11-14:
I think I may be having the same issue deploying from the ESPHome integration in Home Assistant:

https://github.com/konnected-io/konnected-esphome/issues/22

Will console into the alarm later and see, mine is also 1.8 Pro

heythisisnate, 2023-11-15:
Add the following to the top of your adopted YAML in the _substitutions_ section:
```
  ethernet_type: RTL8201
```

For an **Alarm Panel Pro v1.8** connecting via **Ethernet** your adopted YAML config should look like this:

```
substitutions:
  name: alarm-panel-pro-XXXXXX
  ethernet_type: RTL8201 # Alarm Panel Pro 1.8 and up
packages:
  konnected.alarm-panel-pro-esp32-ethernet: github://konnected-io/konnected-esphome/alarm-panel-pro-esp32-ethernet.yaml@master
esphome:
  name: ${name}
  name_add_mac_suffix: false

api:
  encryption:
    key: XXXXXX

# add any customizations here
```

closed 2023-11-15 by heythisisnate

wjwnz, 2023-11-15:
Perfect!

[07:14:06][C][ethernet:222]: Ethernet:
[07:14:06][C][ethernet:354]:   IP Address: 192.168.3.46
[07:14:06][C][ethernet:355]:   Hostname: 'alarm-panel-pro-512738'
[07:14:06][C][ethernet:356]:   Subnet: 255.255.255.0
[07:14:06][C][ethernet:357]:   Gateway: 192.168.3.1
[07:14:06][C][ethernet:362]:   DNS1: 192.168.3.1
[07:14:06][C][ethernet:363]:   DNS2: 0.0.0.0
[07:14:06][C][ethernet:383]:   MAC Address: C0:49:EF:51:27:3B
[07:14:06][C][ethernet:388]:   Is Full Duplex: YES
[07:14:06][C][ethernet:393]:   Link Speed: 100
[07:14:06][C][ethernet:227]:   MDC Pin: 16
[07:14:06][C][ethernet:228]:   MDIO Pin: 17
[07:14:06][C][ethernet:229]:   Type: RTL8201
[07:14:06][C][ethernet:230]:   PHY addr: 0