konnected-io/konnected-esphome #17: Garage Close Action should not Open the Garage Door

networkneil, 2023-10-20:
Currently the Closed Button in the web portal and when linked to Apple Homebridge integration, for Home Assistant, the Close Action Button is opening the Door, this breaks the Good Night Scenes I have created in Apple Home.

I'm using just the Range Sensors in my garage setup.

Please help me figure this out and create a better close action which checks if its already closed and will not trigger any action, by which I mean is dont open when the action is to close but confirm that its closed.

This is my yaml config.

```
substitutions:
  name: garage-door-6f6910
  friendly_name: Garage Door
packages:
  konnected.garage-door-esp8266: github://konnected-io/konnected-esphome/garage-door-esp8266.yaml@master
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.22.0.131
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.22.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
```

cecilkootz, 2024-01-20:
I just added my garage doors to presence automation and experienced the same. Did you figure this out?

networkneil, 2024-01-20:
I was able to fix it by adding some additional logic to the close action in cover definition.

`cover:
- platform: template
  name: $garage_door_cover_name
  device_class: garage
  lambda: |-
      if (id(garage_door_range_sensor).state) {
        return COVER_OPEN;
      } else {
        return COVER_CLOSED;
      }
  open_action:
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button
  close_action:
    _**- if:
        condition:
          and:
            - binary_sensor.is_on: garage_door_range_sensor**_
        then:
          - light.turn_on:
              id: warning_led
              effect: strobe
          - light.turn_on:
              id: warning_beep
              effect: strobe
          - delay: $garage_door_close_warning_duration
          - light.turn_off: warning_led
          - light.turn_off: warning_beep
          - button.press: garage_door_opener_button
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - light.turn_on:
              id: warning_led
              effect: strobe
          - light.turn_off: warning_led

  stop_action:
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button
    - if:
        condition:
          and:
            - binary_sensor.is_on: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button`
          
Hope this helps!

networkneil, 2024-01-20:
I was able to fix it by adding some additional logic to the close action in cover definition.

```
cover:
- platform: template
  name: $garage_door_cover_name
  device_class: garage
  lambda: |-
      if (id(garage_door_range_sensor).state) {
        return COVER_OPEN;
      } else {
        return COVER_CLOSED;
      }
  open_action:
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button
  close_action:
    - if:
        condition:
          and:
            - binary_sensor.is_on: garage_door_range_sensor
        then:
          - light.turn_on:
              id: warning_led
              effect: strobe
          - light.turn_on:
              id: warning_beep
              effect: strobe
          - delay: $garage_door_close_warning_duration
          - light.turn_off: warning_led
          - light.turn_off: warning_beep
          - button.press: garage_door_opener_button
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - light.turn_on:
              id: warning_led
              effect: strobe
          - light.turn_off: warning_led

  stop_action:
    - if:
        condition:
          and:
            - binary_sensor.is_off: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button
    - if:
        condition:
          and:
            - binary_sensor.is_on: garage_door_range_sensor
        then:
          - button.press: garage_door_opener_button
```
  close_action:
    **_- if:
        condition:
          and:
            - binary_sensor.is_on: garage_door_range_sensor_**
        then:
          - light.turn_on:
              id: warning_led
              effect: strobe
          - light.turn_on:
              id: warning_beep
              effect: strobe
          - delay: $garage_door_close_warning_duration
          - light.turn_off: warning_led
          - light.turn_off: warning_beep
          - button.press: garage_door_opener_button

Hope this helps!

danielgoepp, 2024-02-28:
Same problem here. I just added a condition in the automation to only close if it is open. But still, this should not be necessary. Close should only close.

networkneil, 2024-02-28:
> Same problem here. I just added a condition in the automation to only close if it is open. But still, this should not be necessary. Close should only close.

If you were able to fix it please post how or please look above I was able fix it. Adding additional logic did the trick.

danielgoepp, 2024-02-29:
Yes, as I said, I just added a condition to my Home Assistant automation.  The details are pretty straight forward, below is my automation. I was mostly just commenting as a +1 for looking into this as I experience it also.

```
alias: Action - Front Door Open - Close Garage Door
description: ""
trigger:
  - type: opened
    platform: device
    device_id: 16b46d6a06370d0b5494c8a8b31103d8
    entity_id: ee22784a86db88905589c81e06fe534f
    domain: binary_sensor
condition:
  - condition: template
    value_template: >
      {{ (as_timestamp(now()) -
      as_timestamp(states.cover.garage_door_car.last_changed)) < 300 }}
  - condition: device
    device_id: b3dafe2e84206270bf5cc723582a6e3e
    domain: cover
    entity_id: bdab316b153f10913ab6aa83d2304a7c
    type: is_open
action:
  - device_id: b3dafe2e84206270bf5cc723582a6e3e
    domain: cover
    entity_id: bdab316b153f10913ab6aa83d2304a7c
    type: close
mode: single
```

networkneil, 2024-02-29:
> Yes, as I said, I just added a condition to my Home Assistant automation. The details are pretty straight forward, below is my automation. I was mostly just commenting as a +1 for looking into this [...]

Got it thank you for sharing. I though it would help others if they are in the same situation.

heythisisnate, 2024-03-05:
This has been fixed in https://github.com/konnected-io/konnected-esphome/commit/cd9f33d90152110a78c6cf626ea5a769c9de7a69

The open and close actions are now state-aware and it will ignore (and play and error tone) if you try to close the garage that's already closed (or vice-versa).

closed 2024-03-05 by heythisisnate

danielgoepp, 2024-03-10:
@heythisisnate this might be by design, but I figured I would share my experience here. Your fix works, but only if the state is fully open or closed. If the door in moving, then triggering open or close will in fact stop the door still. For example, I have an automation that closes the garage door when I open my front door. If the door is already closing (but not fully closed), opening my front door will stop it from closing, and start opening. I can fix this in the automation, as I have. However since you did fix the code, I might recommend taking it one step further which is to detect "close" or "closing" both as states. And if in "closing" to ignore the close action request. Same for open / opening of course. Love the product though, just a little feedback.

heythisisnate, 2024-03-10:
@danielgoepp good feedback, I will consider this for future enhancement. The tricky part is reliably detecting the _opening_ and _closing_ states. Since we're usually relying on just one sensor to know the state of the garage, the transition states kinda have to be assumed or implied, so it won't always be perfect. I have some ideas to improve this.

danielgoepp, 2024-03-10:
@heythisisnate if I recall correctly the new version supports this better right? The ones coming out in like a week? They get the door status from the unit instead of needing sensors? I can't wait :)