konnected-io/konnected-esphome #21: Failed Config:  Error: Couldn't find ID 'gdo_err_margin'

titelies, 2023-11-12:
So, this issue started happening yesterday, November 11th. I can no longer run a Validate or Install cause I get this error:

INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/garage-door-6f65e9.yaml...
Failed config

binary_sensor.template: [source /data/packages/d9505d5d/packages/vl53l0x-range-sensor.yaml:44]
  platform: template
  id: garage_door_range_sensor
  internal: True
  
  Couldn't find ID 'gdo_err_margin'. Please check you have defined an ID with that name in your configuration.
  lambda: !lambda |-
    float calibrated_distance = id(open_garage_door_distance_from_ceiling).state;
    if (std::isnan(id(range_sensor).state) || id(range_sensor).state > calibrated_distance + id(gdo_err_margin)) {
      return false;
    } else if (id(range_sensor).state > calibrated_distance - id(gdo_err_margin)){
      return true;
    } else {
      return {};
    }
  filters: 
    - delayed_off: 3s
  disabled_by_default: False
  name: garage_door_range_sensor

[image]

heythisisnate, 2023-11-13:
This was fixed in c0a5ff09d76d044a019fc7cdf60a08c63891aadf

closed 2023-11-13 by heythisisnate

titelies, 2023-11-13:
I believe that's when the issue started, I had no issues on Friday. Can you provide any other guidance... 
I used the Konnected Garge Door Opener Firmware website to install the 0.4.2 Firmware and the issue still persists.

heythisisnate, 2023-11-13:
Probably ESPHome is caching an older version of the package. Try removing the `.esphome/packages` folder to force it to re-download from Github

titelies, 2023-11-13:
> Probably ESPHome is caching an older version of the package. Try removing the `.esphome/packages` folder to force it to re-download from Github

Where do I find this packages directory in Home Assistant? Google can't even seem to help me...

heythisisnate, 2023-11-13:
Try clicking on "Clean build files" in the ESPHome dashboard.
The ESPHome build cache should be in `/data/.esphome` if you've installed as a HA add-on.

titelies, 2023-11-13:
> Try clicking on "Clean build files" in the ESPHome dashboard. The ESPHome build cache should be in `/data/.esphome` if you've installed as a HA add-on.

Thanks for trying. Clean Build files wasn't working but I found the changes I had to make to the File Editor Add-on configuration in order to be able to find the .storage directory and I deleted the file and then ran the Validate and Install successfully. 
Have a good week!