konnected-io/konnected-esphome #24: Warnings in HA when running

michael-robbins, 2023-11-19:
Looks like HA is printing these warnings for the alarm panel pro and the garage opener (same message).

```
2023-11-19 10:31:28.264 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.alarm_panel_pro_xxxxxx_wifi_signal (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

Should I raise a HA core bug as it says? Or is this more here where we're reporting the wrong value/type?

danielgoepp, 2023-11-19:
Your timing is great @michael-robbins - I was just looking at this same warning this morning :) I have it for the garage doors:

```
2023-11-19 03:47:31.346 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.garage_door_opener_wifi_signal_4 (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-11-19 03:47:31.723 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.garage_door_opener_wifi_signal_2 (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

heythisisnate, 2023-11-19:
It's addressed here: https://github.com/esphome/issues/issues/4763

It's just a warning, so don't worry about it. The ESPHome documentation and Home Assistant didn't agree on valid units for signal_strength, so it looks like the documentation was updated. I will update the Konnected packages accordingly.

si458, 2024-04-15:
@heythisisnate have u updated the konnected esp yamls at all for the blank device_class?

danielgoepp, 2024-06-16:
Just an FYI on this one. @heythisisnate your comment on 11/19/23 says this is fixed in 4763, but I'm still seeing these messages. Not a big deal necessarily, seems to work, but would be nice to not get this warning. I'm running the latest HA (2024.6.3) and the latest blaQ (1.2.0).

danielgoepp, 2024-06-24:
I'm also now getting an additional message that looks very related, but is slightly different, having to do with Recorder.

```
2024-06-24 04:40:10.557 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.garage_door_opener_workshop_wifi_signal (dBm) cannot be converted to the unit of previously compiled statistics (%). Generation of long term statistics will be suppressed unless the unit changes back to % or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2024-06-24 04:40:10.557 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.garage_door_opener_car_wifi_signal (dBm) cannot be converted to the unit of previously compiled statistics (%). Generation of long term statistics will be suppressed unless the unit changes back to % or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2024-06-24 04:40:10.808 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.garage_door_opener_car_wifi_signal_2 (%) cannot be converted to the unit of previously compiled statistics (dBm). Generation of long term statistics will be suppressed unless the unit changes back to dBm or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

closed 2025-08-01 by heythisisnate

danielgoepp, 2025-08-02:
Thanks @heythisisnate! Appreciate the update.

commit 96a57042 (2025-08-01): Update WiFi Signal percent device class to silence warnings; fixes #24