konnected-io/konnected-esphome #27: Additional substitution to set allow_other_uses to false for alarm pin

jcbshw, 2024-01-01:
Would it be possible to add a substitution to set allow_other_uses to false for the alarm pin?

Currently there is only the substitution for the warning beep pin. My alarm pin is only used for the alarm so it currently fails to build.

```
  Pin 15 incorrectly sets allow_other_uses: true.
  platform: gpio
  id: siren
  name: Siren
  pin: 
    number: 15
    allow_other_uses: True
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  icon: mdi:bullhorn
  disabled_by_default: False
  restore_mode: ALWAYS_OFF
  interlock_wait_time: 0ms
```

zorkakk, 2024-01-06:
Hi! I have the same or at least similar problem. In one of my alarm panels, I am not using the alarm pin or the out pin and I still get the same error. I also have a problem with another alarm panel where I am using the out pin and it is mentioned in the code only once but the error says that I am using it in multiple places. My knowledge is too basic to know what the problem is. I cannot add allow_other_uses because it is not allowed for the switch.gpio

SteeledKnight, 2024-01-07:
I have the same issue. You'll get this issue if you do not have something additional set to the alarm's output (out). By default the warning beep is mapped to "out". I changed the warning beep to another pin that I have connected to a piezo buzzer. I just forked repository and made the necessary change. See my pull request, #28, if you want to see the details.

Clasper101, 2024-03-23:
I too was having issues with this as I wanted to use zone5 for the piezo buzzer. I followed the Konnected instructions for changing 'out' to zone 5, but the the yaml had 2 issue as shown below:
####
  # CONNECTION MAPPINGS
  warning_beep_pin: $zone5

[image]
.
The second one, pin 13 was resolved with the substitution 'warning_beep_shared: "false"', but the pin 15 issue took a bit more work, as a substitution wouldn't work. Instead I copied the alarm yaml code from Github, and placed it in my yaml file, and changed the allow other uses to false, then I deleted the request for downloading the alarm yaml file fro Github. My yaml the validated ok.
 This package is required and sets up core features.
      - packages/core-esp8266.yaml
      - packages/alarm-panel/zone1.yaml
      - packages/alarm-panel/zone2.yaml
      - packages/alarm-panel/zone3.yaml
      - packages/alarm-panel/zone4.yaml
      - packages/alarm-panel/zone6.yaml
             
      ####
There is other code here, I pasted it at the end.
switch:
  - id: siren
    name: Siren
    pin: 
      number: $alarm
      allow_other_uses: false
    platform: gpio
    icon: mdi:bullhorn 

A few hours of thoughts, research and trials.

I hope this helps.