konnected-io/konnected-esphome #32: Unable to connect to Alarm Panel Pro

rayzorben, 2024-02-13:
Using v1.7 Alarm Panel Pro

I connected a USB cable to my Linux computer, and to the alarm panel, and tried flashing, but it could not find it.

I got a windows laptop and installed ALL 3 sets of drivers and tried 2 different cables, and it still could not detect the Alarm Panel Pro.

Any ideas?

SteeledKnight, 2024-02-14:
What browser are you using?

rayzorben, 2024-02-14:
Microsoft Edge

SteeledKnight, 2024-02-14:
Have you tried another browser? I had issues with Firefox on Linux. Had to use Chrome

rayzorben, 2024-02-14:
@SteeledKnight did you have to install any special drivers on linux or did it just work?

rayzorben, 2024-02-14:
Tried Chrome too, no luck. I did a `journalctl --follow` thinking it would recognize something being plugged in but nothing is logged when I do.

The Alarm Panel Pro is powered by POE, and I am plugging into the MicroUSB port with a cable that I am certain is designed for data. 

Does it need to be plugged in with normal power and MicroUSB?

SteeledKnight, 2024-02-14:
I know the Linux kernel ships with the needed drivers, so drivers shouldn't be needed on Linux. I hate to ask, but have you confirmed with another device that the cable carries data? I went through 5 cables before I found one with data pins connected. It might be worth connecting to DC power if you can

rayzorben, 2024-02-14:
The website for Konnected says PoE or DC. But no, I am not sure. You know how may devices I have that still use MicroUSB for data? ZERO.

heythisisnate, 2024-02-14:
On linux it should show up as `/dev/ttyUSB` or similar. Do you see any ports that look like that when plugged in?

rayzorben, 2024-02-14:
> On linux it should show up as `/dev/ttyUSB` or similar. Do you see any ports that look like that when plugged in?

No. I just ordered a new cable from Amazon to make sure I have the correct data cable.

closed 2024-04-04 by heythisisnate