konnected-io/konnected-esphome #33: ESPHome 2024.2.2 update error on Garage

tasi333, 2024-03-06:
I get the following error when I try to update Konnected Garage to the latest ESPHome 2024.2.2 on Home Assistant:

INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/garage-door-59c0cb.yaml...
Failed config

binary_sensor.unknown: [source /data/packages/d9505d5d/packages/garage-door-cover-wired.yaml:4]
  
  Source for extension of ID 'garage_door_input' was not found.
  id: !extend garage_door_input
  on_state: 
    - if: 
        condition: 
          lambda: return x;
        then: 
          - cover.template.publish: 
              id: garage_door
              state: OPEN
              current_operation: IDLE
        else: 
          - cover.template.publish: 
              id: garage_door
              state: CLOSED
              current_operation: IDLE

heythisisnate, 2024-03-06:
Please see release notes here regarding the order of packages:
https://github.com/konnected-io/konnected-esphome/releases/tag/2024.3.0

I suspect all you need to do is ensure that `packages/garage-door-wired-sensor.yaml` is included before the `cover` package.

tasi333, 2024-03-07:
Thanks @heythisisnate, that seems to have worked, thanks for the quick response.

closed 2024-03-07 by tasi333