konnected-io/konnected-esphome #38: no siren on webserver

techunleadit, 2024-03-27:
Hello everyone,
after updating to version 2024.3.2 of alarm-panel-esp8266, some information including the siren switch disappeared from the webserver.
Here is an alarm-panel-esp8266 with the latest firmware
[image]

And here is an alarm-panel-esp8266 with an old version
[image]

From Home Assistant, as always, it is not possible to get information regarding the firmware version, however, the switches for the siren remained.
[image]

heythisisnate, 2024-04-25:
I have not been able to reproduce this issue. Is this a fresh install from install.konnected.io? Have you customized the firmware in any way?

The siren is added in the `alarm.yaml` package included here: https://github.com/konnected-io/konnected-esphome/blob/master/alarm-panel-esp8266.yaml#L91

techunleadit, 2024-04-26:
Here the config from ESPHome by HomeAssistant

`####
##
## Konnected Alarm Panel (ESP8266)
## Firmware configuration for ESPHome
##
## filename: alarm-panel-esp8266.yaml
## GitHub: https://github.com/konnected-io/konnected-esphome
## Buy Konnected hardware: https://konnected.io
## Help & Support: https://help@konnected.io (support is provided for purchasers of Konnected hardware)
##
## Copyright© 2023 Konnected Inc.
##
## Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation 
## files (the "Software"), to deal in the Software without restriction, including without limitation the rights to use, copy, 
## modify, merge, publish, distribute, sublicense, and/or sell copies of the Software, and to permit persons to whom the Software
## is furnished to do so, subject to the following conditions:
##
## The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
##
## THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
## OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE 
## LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR 
## IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
##
####

####
# GENERAL SETTINGS
# Customize these variables to your preferences and needs
# more: https://esphome.io/guides/configuration-types.html#substitutions
substitutions:

  ####
  # NAME
  # By default, the name of the ESPHome device is "konnected-xxxxxx" where xxxxxx is a unique identifier. The device's
  # hostname on your network is also defined by the name, defaulting to "konnected-xxxxxx.local". Edit this variable to
  # customize the name and hostname. Note: only lowercase characters, numbers and hyphen(-) are allowed.
  name: alarm-panel-9bc338
  friendly_name: Konnected Alarm Panel P1-1
  project_name: konnected.alarm-panel-esp8266
  project_version: "1.0.0"

  ####
  # SETTINGS
  sensor_debounce_time: 200ms
  warning_beep_pulse_time: 100ms
  warning_beep_pause_time: 130ms
  warning_beep_internal_only: "false"
  warning_beep_shared: "true"
  blink_on_state: "true"

  ####
  # ZONE MAPPING
  # Do not edit this section.
  zone1: D1
  zone2: D2
  zone3: D5
  zone4: D6
  zone5: D7
  zone6: D9
  alarm: D8
  out: D8
  status_led: D4
  
  ####
  # CONNECTION MAPPINGS
  warning_beep_pin: $out

####
# PACKAGES
# Each package includes an alarm panel feature
# Remove or comment out any packages that you do not need or want.
packages:

  remote_package:
    url: http://github.com/konnected-io/konnected-esphome
    ref: master
    refresh: 5min
    files:

      ####
      # CORE
      # This package is required and sets up core features.
      - packages/core-esp8266.yaml
      - packages/alarm-panel/zone1.yaml
      - packages/alarm-panel/zone2.yaml
      - packages/alarm-panel/zone3.yaml
      - packages/alarm-panel/zone4.yaml
      - packages/alarm-panel/zone5.yaml
      - packages/alarm-panel/zone6.yaml
      - packages/alarm-panel/alarm.yaml
    
      ####
      # WIFI
      # Enables WiFi connectivity and diagnostics.
      - packages/wifi.yaml

      ####
      # WARNING BEEP
      # Enables a 'Warning Beep' entity, intended to be used with a piezo buzzer. This is 
      # implemented using a light entity with strobe effect to create a repeated beeping sound.
      - packages/warning-beep.yaml

      ####
      # STATUS LED
      # Enables the onboard blue status LED as an activity/error indicator
      - packages/status-led.yaml

dashboard_import:
  package_import_url: github://konnected-io/konnected-esphome/alarm-panel-esp8266.yaml@master
  import_full_config: false

####
# WEB SEVER
# Enables the built-in web server for viewing the device state, internals and controls via web browser
# on the same local network as the device.
web_server:
  include_internal: true

#### 
# LOGGER
# more: https://esphome.io/components/logger.html
logger:
  level: VERBOSE

####
# NATIVE API for HOME ASSISTANT
# Enables the native API for Home Assistant
# more: https://esphome.io/components/api.html
api:
  encryption:
    key: ciGhrerY79Pine56GSCWETWsJxOMSOp423E1Iw7OAPM=

#### 
# OTA UPDATES
# Enables over-the-air updates
# more: https://esphome.io/components/ota.html
ota:`

heythisisnate, 2024-04-26:
I don't see anything wrong with the ESPHome config (warning: your encryption key was exposed).

From the original screenshot it looks as if half of the web page is cut-off. The entities should be listed there in alphabetical order but it is starting at "Uptime" and excluding everything above it including Device ID, IP address, etc. Maybe some error with the web server? Are there errors in the browser console when you load the page?

techunleadit, 2024-04-27:
no errors at all, seems to not load the remaining entities. I've tried to compile it again via ESPHome from HomeAssistant but after adopting it keeps wanting to adopt it again, although the compilation and installation of the firmware is successful.

heythisisnate, 2024-04-28:
I've tried compiling a firmware from the YAML config file that you shared, and it's working fine for me. I don't know what's going on here, but it doesn't seem to be a Konnected config problem.
[image]

closed 2024-04-28 by heythisisnate

techunleadit, 2024-04-29:
Hi Nate, tried today (I saw that I had version 1.0.0 and not the latest 1.0.3) but the situation is still the same.
I compile from ESPHome, upload directly to Konnected and it is successful, but showing the usual page with the usual sensors.
Does it change anything compared to install.konnected.io?

heythisisnate, 2024-04-29:
The firmware distributed on install.konnected.io is built directly from the same config YAML in this repo, so it shouldn't be any different. I don't think this problem is firmware related at all. It seems like a network or web server issue. I have actually seen this a couple times when there's a poor network connection, but usually a page refresh will fix it.

The web-page is constructed client-side with data from the event stream. Can you open the Chrome Inspector > Network > Fetch/XHR > look for the _events_ request > EventStream. Then refresh the page. Here we should see the states initializing:

[image]

techunleadit, 2024-04-29:
Hi have this:

```
id: 15392370
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392371
event: log
data: [0;37m[V][json:058]: Size after shrink 116 bytes[0m

retry: 30000
id: 15392372
event: ping
data: {"title":"Konnected Alarm Panel P1-1 9bc338","comment":"","ota":true,"log":true,"lang":"en"}

id: 15392378
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392383
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_1","name":"Zone 1","icon":"","entity_category":0,"value":false,"state":"OFF"}

id: 15392392
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392395
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_2","name":"Zone 2","icon":"","entity_category":0,"value":false,"state":"OFF"}

id: 15392402
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392407
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_3","name":"Zone 3","icon":"","entity_category":0,"value":false,"state":"OFF"}

id: 15392414
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392419
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_4","name":"Zone 4","icon":"","entity_category":0,"value":false,"state":"OFF"}

id: 15392426
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392430
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_5","name":"Zone 5","icon":"","entity_category":0,"value":false,"state":"OFF"}

id: 15392438
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392442
event: log
data: [0;37m[V][json:058]: Size after shrink 120 bytes[0m

event: state
data: {"id":"binary_sensor-zone_6","name":"Zone 6","icon":"","entity_category":0,"value":true,"state":"ON"}

id: 15392461
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392463
event: log
data: [0;37m[V][json:058]: Size after shrink 208 bytes[0m

event: state
data: {"id":"light-warning_beep","name":"Warning Beep","icon":"","entity_category":0,"state":"OFF","effect":"None","color_mode":"onoff","color":{},"effects":["None","Strobe"]}

id: 15392476
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392481
event: log
data: [0;37m[V][json:058]: Size after shrink 168 bytes[0m

event: state
data: {"id":"light-blue_status_led","name":"blue_status_led","icon":"","entity_category":0,"state":"OFF","color_mode":"onoff","color":{},"effects":["None"]}

id: 15392500
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392502
event: log
data: [0;37m[V][json:058]: Size after shrink 156 bytes[0m

event: state
data: {"id":"sensor-uptime","name":"Uptime","icon":"mdi:timer-outline","entity_category":2,"value":15359.03,"state":"15359 s","uom":"s"}

id: 15392508
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392513
event: log
data: [0;37m[V][json:058]: Size after shrink 152 bytes[0m

event: state
data: {"id":"sensor-wifi_signal_rssi","name":"WiFi Signal RSSI","icon":"","entity_category":2,"value":-58,"state":"-58 dBm","uom":"dBm"}

id: 15392553
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

id: 15392641
event: log
data: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m

retry: 30000
id: 15397745
event: ping
data:
```

heythisisnate, 2024-04-29:
I don't see any real problem here. It looks like the firmware is working fine.

There's a warning about the API connection. Is 192.168.4.3 your Home Assistant server? I think `errno=11` means something to to with the API encryption. Maybe the encryption key is wrong or not set in Home Assistant. However this should be irrelevant to the problem you're facing with the web page.

Maybe the verbose logging is overwhelming your browser? Try changing the logger level to `DEBUG` instead of `VERBOSE` and re-install.