konnected-io/konnected-esphome #41: Rename Zones and IDs in ESPHome code

livetoautomate, 2024-04-04:
I migrated from the stock firmware to ESPHome without too much trouble.  

After making the switch, I realized using the Konnected ESPHome code in this repo doesnt (easily) allow me to modify the name of the binary_sensor. They come in as Zone1-Zone6 due to the Repo ZoneX.yaml files.

I know I can change it in the [ESPHome HA integration([](https://support.konnected.io/rename-zones-in-home-assistant)), but I prefer to set it at the source (in the ESP code).  This way, if I ever have an issue, they will always come in the same and I don't ever have to reset them in HA.

What are your thoughts about updating the alarm-panel\zoneX.yaml files to Variablize the zone id and names...? 

EX: 
binary_sensor:
  - id: $zone1
    name: $zone1

heythisisnate, 2024-04-04:
There are two ways to solve this:

1. If you want to manage the ESPHome config YAML yourself -- use the _!extend_ keyword to add additional config options to your zones or other entities. See: https://support.konnected.io/esphome-customize-zone-inputs-on-konnected-alarm-panel

2. If managing YAML files and learning ESPHome config syntax is not for you -- we've built a firmware customization service that is orchestrated by the Konnected app. Download the Konnected app for iOS or Android, use it to discover your Konnected device running ESPHome, then you will see a series of firmware settings screens where you can customize zone names, types, etc and a variety of other settings. When done, our platform will automatically compile an updated firmware for your device with your settings and preferences already built-in.

closed 2024-04-04 by heythisisnate

livetoautomate, 2024-04-04:
Thanks @heythisisnate.  Option One is exactly what I was looking for I just missed it in the Konnected docs.