konnected-io/konnected-esphome #47: Unable to flash ESPHome

wizbowes, 2024-04-26:
Following the instructions here: https://support.konnected.io/esphome-customization-on-the-konnected-alarm-panel-pro-esp32, updating using the following code:

```
substitutions:
  name: konnected-f39e30
  friendly_name: Alarm Panel
packages:
  remote_package:
    url: http://github.com/konnected-io/konnected-esphome
    ref: master
    refresh: 5min
    files:
  
      # Required packages:
      - packages/alarm-panel-esp32-base.yaml
      - packages/status-led.yaml      

      # Keep either the wifi.yaml or ethernet.yaml package below (not both)
      - packages/wifi.yaml

      # Default input/output configuration
      - packages/alarm-panel/zone1.yaml
      - packages/alarm-panel/zone2.yaml
      - packages/alarm-panel/zone3.yaml
      - packages/alarm-panel/zone4.yaml
      - packages/alarm-panel/zone5.yaml
      - packages/alarm-panel/zone6.yaml
      - packages/alarm-panel/zone7.yaml
      - packages/alarm-panel/zone8.yaml
      - packages/alarm-panel/zone9.yaml
      - packages/alarm-panel/zone10.yaml
      - packages/alarm-panel/zone11.yaml
      - packages/alarm-panel/zone12.yaml
      - packages/alarm-panel/alarm1.yaml
      - packages/alarm-panel/alarm2.yaml
      - packages/warning-beep.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <removed>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
logger:
ota:

```

When I try to update OTA I get the following:

```
INFO ESPHome 2024.3.1
INFO Reading configuration /config/konnected-f39e30.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing konnected-f39e30 (board: wesp32; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
*** [.pioenvs/konnected-f39e30/src/esphome/components/dashboard_import/dashboard_import.o] Source `src/esphome/components/dashboard_import/dashboard_import.cpp' not found, needed by target `.pioenvs/konnected-f39e30/src/esphome/components/dashboard_import/dashboard_import.o'.
========================== [FAILED] Took 3.05 seconds ==========================
```

How to proceed?

techunleadit, 2024-04-29:
Hi, I'll see if I can help you...
I assume you want to use Konnected Pro in Ethernet mode, use this example yaml:
https://github.com/konnected-io/konnected-esphome/blob/master/alarm-panel-pro-esp32-ethernet.yaml
In case you want to use the WiFi version, here is the other file:
https://github.com/konnected-io/konnected-esphome/blob/master/alarm-panel-pro-esp32-wifi.yaml

wizbowes, 2024-04-29:
That fixed it.  Thanks,

closed 2024-04-29 by wizbowes