konnected-io/konnected-esphome #57: GDO blaQ / purple button / Home Assistant: light entity state incorrect after manually changing light

drothenberger, 2024-05-16:
Steps to Reproduce:
1. Open the garage door. Notice that the light on the GDO turns on, and the light entity in Home Assistant reflects this.
2. Use the Home Assistant entity to turn off the light. Note that it turns off on the GDO correctly.
3. Close the garage door.

At this point, the light on the GDO turns on (correct), but the Home Assistance entity stays off. Further opening and closing of the GDO does turn on the actual light, but the HA entity still remains off.

This is with ESPHome 2024.5.0 and version 1.1.1 of the firmware.

drothenberger, 2024-06-17:
I am also seeing the same RX data size errors as reported in #63.

danielgoepp, 2024-06-18:
In addition to the error I reported on #63, I'm also getting a similar error for locking. For what it's worth, this is the output of the logs:

```
[09:08:41][W][web_server_idf:070][httpd]: Only application/x-www-form-urlencoded supported for POST request
[09:08:41][D][lock:054]: 'Lock': Sending state UNLOCKED
[09:08:41][V][esp-idf:000][gdo_main_task]: E (2875014) gdolib: Parity error, check wiring?
[09:08:41][V][esp-idf:000][gdo_main_task]: E (2875105) gdolib: RX data size error: 3
[09:08:42][V][esp-idf:000][gdo_main_task]: E (2875525) gdolib: Parity error, check wiring?
[09:08:42][V][esp-idf:000][gdo_main_task]: E (2875615) gdolib: RX data size error: 3
[09:08:42][I][secplus_gdo:056][gdo_main_task]: Lock: Unlocked
[09:08:47][W][web_server_idf:070][httpd]: Only application/x-www-form-urlencoded supported for POST request
[09:08:47][D][lock:054]: 'Lock': Sending state LOCKED
[09:08:47][V][esp-idf:000][gdo_main_task]: E (2881145) gdolib: RX data size error: 3
[09:08:47][V][esp-idf:000][gdo_main_task]: E (2881293) gdolib: Parity error, check wiring?
[09:08:48][I][secplus_gdo:063][gdo_main_task]: Door: Open, 100%, target: 100%
[09:08:48][D][cover:170][gdo_main_task]: 'Garage Door' - Publishing:
[09:08:48][D][cover:173][gdo_main_task]:   Position: 100%
[09:08:48][D][cover:186][gdo_main_task]:   Current Operation: IDLE
[09:08:48][V][esp-idf:000][gdo_main_task]: E (2881554) gdolib: Parity error, check wiring?
[09:08:48][I][secplus_gdo:056][gdo_main_task]: Lock: Locked
[09:08:49][I][secplus_gdo:063][gdo_main_task]: Door: Closed, 0%, target: 100%
[09:08:49][D][cover:170][gdo_main_task]: 'Garage Door' - Publishing:
[09:08:49][D][cover:173][gdo_main_task]:   Position: 0%
[09:08:49][D][cover:186][gdo_main_task]:   Current Operation: IDLE
```

danielgoepp, 2024-06-18:
Also to clarify my experiences vs what @drothenberger reported, I'm just clicking the turn light on/off button in the web interface (or lock / unlock), not Home Assistant.

danielgoepp, 2024-06-21:
I just tested the light again. Interestingly now the light does go on and off, but I still get errors:

```
[08:50:06][W][web_server_idf:070][httpd]: Only application/x-www-form-urlencoded supported for POST request
[08:50:06][D][light:036]: 'Garage Light' Setting:
[08:50:06][D][light:047]:   State: ON
[08:50:06][V][mqtt:477]: Publish(topic='garage-door-opener-car/light/garage_light/state' payload='{"color_mode":"onoff","state":"ON","color":{}}' retain=1 qos=0)
[08:50:06][V][esp-idf:000][gdo_main_task]: W (110522) gdolib: Sent command: TOGGLE_LIGHT_PRESS
[08:50:06][V][esp-idf:000][gdo_main_task]: E (110671) gdolib: RX data size error: 3
[08:50:07][V][mqtt:477]: Publish(topic='garage-door-opener-car/sensor/wifi_signal__/state' payload='100' retain=1 qos=0)
[08:50:07][V][mqtt:477]: Publish(topic='garage-door-opener-car/sensor/wifi_signal/state' payload='-49' retain=1 qos=0)
[08:50:07][I][secplus_gdo:052][gdo_main_task]: Light: On
[08:50:07][V][mqtt:477]: Publish(topic='garage-door-opener-car/light/garage_light/state' payload='{"color_mode":"onoff","state":"ON","color":{}}' retain=1 qos=0)
[08:50:07][V][esp-idf:000][gdo_main_task]: W (111026) gdolib: Sent command: TOGGLE_LIGHT_RELEASE
[08:50:07][V][esp-idf:000][gdo_main_task]: E (111114) gdolib: Parity error, check wiring?
[08:50:07][V][esp-idf:000][gdo_main_task]: E (111205) gdolib: RX data size error: 3
[08:50:10][W][web_server_idf:070][httpd]: Only application/x-www-form-urlencoded supported for POST request
[08:50:10][D][light:036]: 'Garage Light' Setting:
[08:50:10][D][light:047]:   State: OFF
[08:50:10][V][mqtt:477]: Publish(topic='garage-door-opener-car/light/garage_light/state' payload='{"color_mode":"onoff","state":"OFF","color":{}}' retain=1 qos=0)
[08:50:10][V][esp-idf:000][gdo_main_task]: W (114107) gdolib: Sent command: TOGGLE_LIGHT_PRESS
[08:50:10][V][esp-idf:000][gdo_main_task]: E (114238) gdolib: RX data size error: 4
[08:50:10][V][esp-idf:000][gdo_main_task]: E (114404) gdolib: Parity error, check wiring?
[08:50:10][I][secplus_gdo:052][gdo_main_task]: Light: Off
[08:50:10][V][mqtt:477]: Publish(topic='garage-door-opener-car/light/garage_light/state' payload='{"color_mode":"onoff","state":"OFF","color":{}}' retain=1 qos=0)
[08:50:10][V][esp-idf:000][gdo_main_task]: W (114612) gdolib: Sent command: TOGGLE_LIGHT_RELEASE
[08:50:10][V][esp-idf:000][gdo_main_task]: E (114620) gdolib: Parity error, check wiring?
[08:50:10][V][esp-idf:000][gdo_main_task]: E (114631) gdolib: Parity error, check wiring?
[08:50:11][V][esp-idf:000][gdo_main_task]: W (114716) gdolib: Unhandled command: 13, resp: fc
[08:50:11][V][mqtt:477]: Publish(topic='garage-door-opener-car/sensor/uptime/state' payload='115' retain=1 qos=0)
```

danielgoepp, 2025-08-01:
Another one on the list of not important, but still not working. I checked in here to see what latest behavior since last upgrading. Right now turning the light on / off does nothing, but it doesn't error either.