konnected-io/konnected-esphome #59: ESPHome/Home Assistant - Sensor gets stuck on opening

omayhemo, 2024-06-02:
Garage door is closed, but the status in home assistant is opening.  I have two, both do the same thing periodically.  Each time the garage door is closed, when the status appears as Opening in Home Assistant.  I've adopted both sensors into ESPHome.

The yaml is as follows:
```
packages:
  konnected.garage-door-esp8266: github://konnected-io/konnected-esphome/garage-door-esp8266.yaml@master
esphome:
  name: garage-door-west
  friendly_name: "Garage Door West"
api:
  encryption:
    key: "REDACTED"

wifi:
  ssid: !secret wifi_ssid_iot
  password: !secret wifi_password
  use_address: REDACTED
  ap:
    ssid: "Garage-Door-West"
    password: "REDACTED"

# Enable logging
logger:

ota:
  password: "REDACTED"

  # Enable fallback hotspot (captive portal) in case wifi connection fails

captive_portal:
   ```

omayhemo, 2024-06-07:
Noone?

JonathanTroyer, 2024-09-24:
Also running into this. It can get stuck on opening _or_ closing. Additionally, when HomeAssistant restarts the sensor defaults back to opening.

JonathanTroyer, 2024-09-27:
I've found a pretty hacky workaround: set up an automation that triggers when the garage is in the opening or closing state for more than 15s and then press the reset button on the device.
[image]