konnected-io/konnected-esphome #69: Homebridge support in homebridge-ratgdo.

hjdhjd, 2024-07-19:
I’d have sent a email but didn’t see a developer contact beyond your support information.

I’m glad to see an additional, intriguing, fork/variant emerge of Ratgdo. I’ve added initial support for your Ratgdo variant to [homebridge-ratgdo](https://github.com/hjdhjd/homebridge-ratgdo). Just wanted to provide a pointer and give the team here a heads up that there’s a robust Homebridge option for those that prefer to access HomeKit support in that way. You’re welcome to provide a pointer to it (or not) as you choose.

Thanks for the contributions to the community!

heythisisnate, 2024-07-19:
We do have a Homebridge plugin now thanks to @KyleBoyer
https://github.com/KyleBoyer/homebridge-blaq

Discussion thread here:
https://community.konnected.io/t/garage-door-opener-homebridge/35594/

closed 2024-07-19 by heythisisnate

hjdhjd, 2024-07-19:
In the interest of trying to keep things simple, I'd encourage the community to focus efforts around a single plugin for Ratgdo variants, but as always, it's open source so you do you. 😄