konnected-io/konnected-esphome #72: ESPHome won't build - see logs

Majestic7979, 2024-09-24:
My yaml passes a validation check.

```
INFO ESPHome 2024.9.1
INFO Reading configuration /config/konnected-bd7c30.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Found and merged multiple configurations for ota platform esphome port(s) [3232]
INFO Generating C++ source...
INFO Compiling app...
Processing konnected-bd7c30 (board: wesp32; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble/ble.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble/ble_advertising.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble/ble_uuid.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_2901.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_2902.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_characteristic.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_descriptor.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_server.o
Compiling .pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_service.o
In file included from src/esphome/components/esp32_ble/ble.h:3,
                 from src/esphome/components/esp32_ble/ble.cpp:3:
src/esphome/components/esp32_ble/ble_advertising.h:9:10: fatal error: esp_bt.h: No such file or directory

****************************************************************
* Looking for esp_bt.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_bt.h"
* Web  > https://registry.platformio.org/search?q=header:esp_bt.h
*
****************************************************************

 #include <esp_bt.h>
          ^~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/esp32_ble/ble_advertising.cpp:1:
src/esphome/components/esp32_ble/ble_advertising.h:9:10: fatal error: esp_bt.h: No such file or directory

****************************************************************
* Looking for esp_bt.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_bt.h"
* Web  > https://registry.platformio.org/search?q=header:esp_bt.h
*
****************************************************************

 #include <esp_bt.h>
          ^~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/esp32_ble/ble_uuid.cpp:1:
src/esphome/components/esp32_ble/ble_uuid.h:9:10: fatal error: esp_bt_defs.h: No such file or directory
[... 115 lines cut ...]
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_2902.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_characteristic.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_descriptor.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_server.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_service.o] Error 1
In file included from src/esphome/components/esp32_ble_server/ble_descriptor.h:3,
                 from src/esphome/components/esp32_ble_server/ble_characteristic.h:3,
                 from src/esphome/components/esp32_improv/esp32_improv_component.h:8,
                 from src/esphome/components/esp32_improv/esp32_improv_component.cpp:1:
src/esphome/components/esp32_ble/ble_uuid.h:9:10: fatal error: esp_bt_defs.h: No such file or directory

*********************************************************************
* Looking for esp_bt_defs.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_bt_defs.h"
* Web  > https://registry.platformio.org/search?q=header:esp_bt_defs.h
*
*********************************************************************

 #include <esp_bt_defs.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_improv/esp32_improv_component.o] Error 1
========================== [FAILED] Took 1.56 seconds ==========================
```

closed 2024-09-24 by heythisisnate

Majestic7979, 2024-09-25:
I did clean the files. It's not building. *** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_characteristic.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble/ble_uuid.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_descriptor.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_2901.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_service.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_improv/esp32_improv_component.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_2902.o] Error 1
*** [.pioenvs/konnected-bd7c30/src/esphome/components/esp32_ble_server/ble_server.o] Error 1

Majestic7979, 2024-09-25:
If the issue is with ESPHome I kindly request that you as the owner of Konnected reach out to them/raise an issue yourself because that way it gives it more weight in terms of urgency as you'll be able to say that this is affecting your whole customer base who are using the esphome-based firmware. I would be grateful if you took ownership of this for us and it shows your commitment to support your product. There's no apparent issue my end and I did clear the build files. It happened after the latest update to ESPHome, I built my Konnected firmware, then the next day I needed to make a slight change but it stopped building, I noticed that ESPHome was updated. So it's likely that it's on their end but I am sure you'll be more effective in getting this resolved for your customers better than me.

heythisisnate, 2024-09-25:
This is a dependency error, not a bug. You can try forcing it to re-download all the dependencies by deleting the `.pioenvs/konnected-bd7c30` directory and then try again.
Or, you can use the Konnected app and we will manage the ESPhome firmware and updates for you.

Majestic7979, 2024-09-25:
Uh, you suggested that I move to esphome (on an issue I opened in Home Assistant), so I won't go back to the Konnected app and firmware. I am disabled and I don't have unlimited time and energy to keep messing with the panel, so I took your suggestion on board and spent considerable time and energy to make the change, I won't return now. I will delete the folder and see what happens, then I'll report back.

Majestic7979, 2024-09-25:
Thanks for the info, I thought the app became redundant once the ESPHome firmware was installed.

So I deleted the folder, and cleaned build files. Same error.

Look at my ESPHome dashboard, I have other devices and only the Konnected board causes this compilation error. That's why I am here...

[image]

Majestic7979, 2024-09-28:
For reference in case any other Konnected customers need help, removing "esp32_improv:" component allowed the code to compile. I also think this is more secure as there's no exposed bluetooth component advertising for setup this way.