konnected-io/konnected-esphome #81: Alarm Panel Pro Eth

Asleep-at-the-wheel, 2025-04-25:
Hello,

I flashed my Konnected 1.5 (connected via Eithernet) with the ESPHome firmware and went to adopt it in HA using the current latest version of ESPHome (Current version: 2025.4.0). The first build and adopt failed with:

Using the current latest version of everything at time of writing:
```
 Core 2025.4.3
Supervisor 2025.04.0
Operating System 15.2
Frontend 20250411.0

```
Compiling .pioenvs/konnected-70ec2c/src/esphome/components/switch/automation.cpp.o
src/esphome/components/mdns/mdns_component.cpp: In member function 'void esphome::mdns::MDNSComponent::compile_records_()':
src/esphome/components/mdns/mdns_component.cpp:65:50: error: 'using element_type = class esphome::api::APINoiseContext' {aka 'class esphome::api::APINoiseContext'} has no member named 'has_psk'
   65 |     if (api::global_api_server->get_noise_ctx()->has_psk()) {
      |                                                  ^~~~~~~
Compiling .pioenvs/konnected-70ec2c/src/esphome/components/switch/switch.cpp.o
*** [.pioenvs/konnected-70ec2c/src/esphome/components/mdns/mdns_component.cpp.o] Error 1
========================= [FAILED] Took 25.89 seconds =========================
```

I then took the template from:
https://github.com/konnected-io/konnected-esphome/blob/master/alarm-panel-pro-esp32-ethernet.yaml

Used that, got the same problem.

After commented out:
```
      - packages/mdns.yaml
```

It compiles and appears to work.

Is the mdns package no longer compatible with ESPHome?

Asleep-at-the-wheel, 2025-04-25:
It all worked fine when I re-flashed and adopted via HA settings rather than using the "Take Control" button in ESPHome. Maybe close the issue if required.

heythisisnate, 2025-04-28:
See #80 

The mdns.yaml package in this repo is new, and was just added last week. It was developed in connection with some enhancements made to ESPHome's built-in mdns component, which have been approved for ESPHome 2025.5.x. Some unrelated changes that were also added to the mdns component in the meantime have caused this compatibility error. The problem will go away once ESPHome 2025.5.x is released.

Note that the packages/mdns.yaml package is not needed if you are not planning on using the Konnected app, so there's no harm in removing it if only using with Home Assistant and managing your own ESPHome settings.