konnected-io/konnected-esphome #84: mdns.yaml:14:14: error: 'USE_WEBSERVER_PORT' was not declared in this scope

ptr727, 2025-05-02:
Looks like the recent [update](https://github.com/konnected-io/konnected-esphome/commit/48055a366d052af32ab283ad159f1159361c9882) adding mdns as a package is causing a build break with esphome 2025.4.1?
 
```
/cache/data/packages/8735b6a9/packages/mdns.yaml: In lambda function:
/cache/data/packages/8735b6a9/packages/mdns.yaml:14:14: error: 'USE_WEBSERVER_PORT' was not declared in this scope
   14 |       txt:
      |              ^                 
```

nickgoh, 2025-05-02:
im having issue compiling too in 2025.4.1

> src/esphome/components/mdns/mdns_component.cpp: In member function 'void esphome::mdns::MDNSComponent::compile_records_()':
src/esphome/components/mdns/mdns_component.cpp:65:50: error: 'using element_type = class esphome::api::APINoiseContext' {aka 'class esphome::api::APINoiseContext'} has no member named 'has_psk'
   65 |     if (api::global_api_server->get_noise_ctx()->has_psk()) {
      |                                                  ^~~~~~~
*** [.pioenvs/alarm-panel-pro-dc604c/src/esphome/components/mdns/mdns_component.cpp.o] Error 1

heythisisnate, 2025-05-02:
@ptr727 thank you for reporting this, the problem is fixed.
@nickgoh your error is addressed in #80

commit 7f2ea402 (2025-05-02): Default to port 0 for MDNS if web_server is not enabled; fixes #84
closed 2025-05-02 by heythisisnate